gazebosim/gz-fuel-tools #73: Command line tools not working for 4.1.0 debs

chapulina, 2020-05-26:
After reinstalling fresh debs on Ubuntu Bionic, I get the following error with the command line tool:

```
$ ign fuel list -t model
Error: Version mismatch. Your configuration file version is
            [4.1.0] but libignition-fuel_tools4.so version is [4.0.0].
```

I only have the 4.1.0 debs installed (not 4.0.0):

<details><summary><pre>apt list --installed "*fuel-tools*"</pre></summary>

```
libignition-fuel-tools/unknown,unknown,now 1.2.0-1~bionic all [installed]
libignition-fuel-tools-dev/unknown,unknown,now 1.2.0-1~bionic all [installed]
libignition-fuel-tools1-1/unknown,now 1.2.0-1~bionic amd64 [installed]
libignition-fuel-tools1-dev/unknown,now 1.2.0-1~bionic amd64 [installed]
libignition-fuel-tools2/unknown,now 2.0.0-1~bionic amd64 [installed]
libignition-fuel-tools2-dev/unknown,now 2.0.0-1~bionic amd64 [installed]
libignition-fuel-tools3/unknown,now 3.2.2-1~bionic amd64 [installed]
libignition-fuel-tools3-dev/unknown,now 3.2.2-1~bionic amd64 [installed]
libignition-fuel-tools4/unknown,now 4.1.0-1~bionic amd64 [installed]
libignition-fuel-tools4-dev/unknown,now 4.1.0-1~bionic amd64 [installed]
```
</details>

That error is coming from here:

https://github.com/ignitionrobotics/ign-fuel-tools/blob/c12f929da26cd5d2648ced0e0b836af134973096/src/cmd/cmdfuel.rb.in#L272-L276

That 4.0.0 is coming from here:

https://github.com/ignitionrobotics/ign-fuel-tools/blob/c12f929da26cd5d2648ced0e0b836af134973096/src/ign.cc#L262-L265

I'm not sure how the debs got to this state. I can't reproduce the issue building from source.

j-rivero, 2020-06-08:
I can not reproduce the problem in a new docker container:

```
root@160d396cab20:/# ign fuel --versions
4.1.0
3.2.2
root@160d396cab20:/# apt list --installed '*fuel-tools*'
Listing... Done
libignition-fuel-tools2/unknown,now 2.0.0-1~bionic amd64 [installed,automatic]
libignition-fuel-tools2-dev/unknown,now 2.0.0-1~bionic amd64 [installed]
libignition-fuel-tools3/unknown,now 3.2.2-1~bionic amd64 [installed,automatic]
libignition-fuel-tools3-dev/unknown,now 3.2.2-1~bionic amd64 [installed]
libignition-fuel-tools4/unknown,now 4.1.0-1~bionic amd64 [installed,automatic]
libignition-fuel-tools4-dev/unknown,now 4.1.0-1~bionic amd64 [installed]
```
My hypothesis is that you could have something installed locally coming from a source building that is conflicting with packages somehow.

chapulina, 2020-06-08:
Thanks for trying, @j-rivero . I just cleaned everything again and I can't reproduce the issue anymore. Sorry for the noise!

closed 2020-06-08 by chapulina