gazebosim/gz-fuel-tools #93: cant build ign-fuel-tools1 on windows: %win_dir% is empty, cant found YAML (yaml-0.1), unknown include path appeared, failed to nmake, etc.

ahmadshiddiqn, 2020-08-02:
* **What's the name and version of the OS you're using?**  
Microsoft Windows [Version 10.0.19041.388]
* **Did the problem start happening recently?** 
It started from the first try of installing gazebo from following the tutorial

Hi, I've been trying build a plugin from source. But it needed ignition-common1 and ignition-fuel-tools4, so I tried to follow the steps to install gazebo on windows using the tutorial steps provided [here](http://gazebosim.org/tutorials?tut=install_on_windows&cat=install)

i managed to follow until step 9, when installin ign-common. I found the workaround and somehow am able to nmake install it. more problem occurs when i was trying to build ign-fuel-tools. 

first of all, when running configure.bat, it expect to download and uzip the required zip files, but it failed (as with the previous steps). here's the output of said line:

```
c:\opt\ign-fuel-tools>call  :download_unzip_install curl-7.57.0-vc15-x64-dll-MD.zip
The system cannot find the batch label specified - download_unzip_install

c:\opt\ign-fuel-tools>call  :download_unzip_install jsoncpp-1.8.4-vc15-x64-dll-MD.zip
The system cannot find the batch label specified - download_unzip_install

c:\opt\ign-fuel-tools>call  :download_unzip_install libyaml-0.1.7-vc15-x64-md.zip
The system cannot find the batch label specified - download_unzip_install

c:\opt\ign-fuel-tools>call  :download_unzip_install libzip-1.4.0_zlip-1.2.11_vc15-x64-dll-MD.zip
The system cannot find the batch label specified - download_unzip_install

c:\opt\ign-fuel-tools>call  :install_ign_project ign-common ign-common1
The system cannot find the batch label specified - install_ign_project
```

the command from the batch file is

```
call %win_lib% :download_unzip_install curl-7.57.0-vc15-x64-dll-MD.zip
call %win_lib% :download_unzip_install jsoncpp-1.8.4-vc15-x64-dll-MD.zip
call %win_lib% :download_unzip_install libyaml-0.1.7-vc15-x64-md.zip
call %win_lib% :download_unzip_install libzip-1.4.0_zlip-1.2.11_vc15-x64-dll-MD.zip
call %win_lib% :install_ign_project ign-common ign-common1
```

i assume both %win_lib% and download_unzip_install are exclusive to the ignition developres' CI system as written in the top of configure.bat. it wasn't that much of an issue as the process still runs.

the second one is on finding all the package, especially the YAML package. I already installed ros, and it cant find everything except  CURL package (its already on rosdeps from installing ros). so i extracted them from the files i downloaded provided in the tutorial. i placed them on rosdeps as where all the library ros was installed (including gazebo). everything except yaml were all found. it is said, it needed yaml-0.1:

```
c:\opt\ign-fuel-tools\build>cmake .. -G "NMake Makefiles" -DCMAKE_INSTALL_PREFIX="" -DCMAKE_BUILD_TYPE="Release" -DBUILD_TESTING:BOOL=False
-- ignition-fuel_tools1 version 1.2.0
-- Operating system is Windows
-- Looking for IgnCURL - found

-- Checking for module 'jsoncpp'
--   Found jsoncpp, version 1.8.4
-- Looking for JSONCPP - found

-- Checking for module 'yaml-0.1'
--   No package 'yaml-0.1' found
-- Could NOT find YAML (missing: YAML_FOUND)
-- Attempting manual search for yaml
-- Looking for yaml headers - not found
-- Could NOT find YAML (missing: YAML_FOUND)
-- Looking for YAML - not found

-- Checking for module 'libzip'
--   Found libzip, version 1.4.0
-- Looking for ZIP - found

-- Looking for ignition-common1 - found

-- Looking for ignition-tools - found

-- BUILD ERRORS: These must be resolved before compiling.
--      Missing: YAML
[... 556 lines cut ...]

at this point, im no longer have knowledge of what happened. even the workaround i tried is something i figured out by reading through the process of what happen in the files and change the line of code without expecting what will be the output. i dont have knowledge of cmake and nmake in the first place, so i dont know what to do anymore.

i dont know if this a bug or this problems just happen to me. i hope i was right by reporting it here. thank you.

in short:

* **What step have I done?**
1. Following [gazebo installation tutorial on windows](http://gazebosim.org/tutorials?tut=install_on_windows&cat=install)
2. Every step on building ign files, the %win_dir% and download_unzip_install is ignored
3. When trying to configure to build (configure.bat) ign-fuel-tools, the YAML package cant be found, so i change it to YAMLCPP and download FindYamlCpp.cmake from [here](https://github.com/tudelft3d/3dfier/blob/master/cmake/Modules/FindYamlCpp.cmake)
4. JSONCPP::JSONCPP include path somehow redirected to an unknown directory (C:/Users/scpeters/Downloads/jsoncpp-1.8.4/install/include), it wasn't on my pc, so I created the directory.
5.  When trying to nmake, it returned error: ZIP::ZIP_NOTFOUND on build.make line 477, so I changed it to zip.lib directory path
6.  Trying again to nmake, return a bunch of warning and error. so i report it here.

* **Step to reproduce**:
1. Install ros1 melodic on windows following [this tutorial](https://wiki.ros.org/Installation/Windows) so that gazebo-9 is installed.
2.  clone ign-common1 branch and ign-fuel-tools4 branch to c:\opt, and do step 10 and 11 on [gazebo installation tutorial on windows](http://gazebosim.org/tutorials?tut=install_on_windows&cat=install), step 11 should produce error.
3. Manually extract jsoncpp-1.8.4-vc15-x64-dll-MD.zip, libyaml-0.1.7-vc15-x64-md.zip, and libzip-1.4.0_zlip-1.2.11_vc15-x64-dll-MD.zip to C:\opt\rosdeps\x64\, try to run configure.bat.
4. download FindYamlCpp.cmake from [here](https://github.com/tudelft3d/3dfier/blob/master/cmake/Modules/FindYamlCpp.cmake), and placed it in C:\opt\rosdeps\x64\lib\cmake\ignition-cmake0\cmake0
5. change `ign_find_package(YAML REQUIRED)` in CMakeLists.txt in ign-fuel-tools repo to `ign_find_package(YAMLCPP REQUIRED)`. Run configure.bat again
6. JSONCPP error, create this C:/Users/scpeters/Downloads/jsoncpp-1.8.4/install/include directory
7. Copy extracted jsoncpp files (from jsoncpp-1.8.4-vc15-x64-dll-MD.zip) to C:/Users/scpeters/Downloads/jsoncpp-1.8.4/install/, and run configure.bat again, this should make configure.bat finish. then, try nmake.
8. an unexpected error should occur from build.make line 477, change all ZIP::ZIP_NOTFOUND to C:\opt\rosdeps\x64\lib\zip.lib including one on line 486. try nmake again
9. the long error and warning should be produced as the nmake progresses.

Camixxx, 2020-10-27:
D:\Gazebo\gz-ws\ign-fuel-tools\build>nmake install

Microsoft (R) 程序维护实用工具 14.16.27043.0 版
版权所有 (C) Microsoft Corporation。  保留所有权利。

[  4%] Linking CXX shared library ignition-fuel_tools1.dll
LINK: command "C:\PROGRA~2\MICROS~1\2017\BUILDT~1\VC\Tools\MSVC\1416~1.270\bin\Hostx86\x64\link.exe /nologo @CMakeFiles\ignition-fuel_tools1.dir\objects1.rsp /out:ignition-fuel_tools1.dll /implib:ignition-fuel_tools1.lib /pdb:D:\Gazebo\gz-ws\ign-fuel-tools\build\src\ignition-fuel_tools1.pdb /dll /version:1.2 /machine:x64 /INCREMENTAL:NO ..\install\Release\bin\libcurl.dll D:\Gazebo\gz-ws\ign-common\build\src\ignition-common1.lib ..\install\Release\lib\jsoncpp.lib ..\install\Release\lib\yaml.lib ..\install\Release\bin\libcurl.dll ..\install\Release\lib\ignition-math4.lib ..\install\Release\lib\jsoncpp.lib ..\install\Release\lib\yaml.lib kernel32.lib user32.lib gdi32.lib winspool.lib shell32.lib ole32.lib oleaut32.lib uuid.lib comdlg32.lib advapi32.lib /MANIFEST /MANIFESTFILE:ignition-fuel_tools1.dll.manifest" failed (exit code 1107) with the following output:
ClientConfig.cc.obj :  MSIL .netmodule or module compiled with /GL found; restarting link with /LTCG; add /LTCG to the link command line to improve linker performance
..\install\Release\bin\libcurl.dll : fatal error LNK1107: invalid or corrupt file: cannot read at 0x350 RRS feed
NMAKE : fatal error U1077: “"C:\Program Files\CMake\bin\cmake.exe"”: 返回代码“0xffffffff”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"”: 返回代码“0x2”
Stop.
NMAKE : fatal error U1077: “"C:\Program Files (x86)\Microsoft Visual Studio\2017\BuildTools\VC\Tools\MSVC\14.16.27023\bin\HostX86\x86\nmake.exe"”: 返回代码“0x2”
Stop.

Camixxx, 2020-10-28:
> D:\Gazebo\gz-ws\ign-fuel-tools\build>nmake install
>
> Microsoft (R) 程序维护实用工具 14.16.27043.0 版
> 版权所有 (C) Microsoft Corporation。 保留所有权利。
>
> [ 4%] Linking CXX shared library ignition-fuel_tools1.dll
> LINK: [...]

I refine it by change the build.make file.
Change every `..\install\Release\bin\libcurl.dll` to `..\install\Release\lib\libcurl_imp.lib`

IliaTraichev, 2020-11-12:
Greetings. For **ahmadshiddiqn** I suggest you start over by deleting the entire build folder in your ign-fuel-tools. Follow the tutorial for Gazebo9 for windows till the .\configure command of step 11. You should have downloaded and extracted all the dependencies from listed in step 2. If the command prompt tells you that you are missing something then go to ..\gz-ws\ign-fuel-tools\build and open the CMakeCache.txt. There find the lines where it had tried to find the missing files and set the path for each of their configurations, includes and libraries. You may need to build ignition-tools as well because it was needed for me. Set the paths for only those it is missing that are causing the build warnings. It is also important that you specify the library paths to the .lib files themselves and set the CMAKE_INSTALL_PREFIX path like shown in the tutorial. Now execute nmake. Hopefully no errors pop up. Then do nmake install. You should have the ign-fuel-tools files build. It worked for me. I wish you luck, ahmadshiddiqn.