gazebosim/gz-fuel-tools #111: Support model dependencies

nkoenig, 2020-08-24:
Currently, a Fuel model is completely self-contained. This can lead to multiple copies of materials and/or meshes. For example, most of the [Cave tiles](https://app.ignitionrobotics.org/search;q=Cave*) use the same texture images. As a result, the disk space used by Fuel models can be large (my directory is sitting at just over 13GB). Docker images that require models to be baked-in (ie no runtime download) can be too large to upload to AWS ECR or Dockerhub.

One way to reduce disk usage it to reference materials from one model inside another. Basically, setup model dependencies.

I believe this is partially supported through the use of URIs. The missing pieces are:

1. When downloading a model, `ign fuel` should also download its dependencies. 
2. The first item may require modifications to `model.config` to support model dependencies. It may be better to parse a newly downloaded model.sdf file and determine what the dependencies are, otherwise the `model.config` file could become out-dated.
3. Gz3d will likely require updates (@nkoenig  can help with this).
4. The Fuel webserver should also track dependencies in order to warn users when they deleted models (@nkoenig can help with this).

JShep1, 2020-10-06:
First PR at getting this issue resolved is here: https://github.com/ignitionrobotics/ign-fuel-tools/pull/123  This should download all models specified in the `<dependency>` tag in the `metadata.pbtxt` file for any given model.

JShep1, 2020-11-10:
Additionally, message conversion for dependencies was added in https://github.com/ignitionrobotics/ign-msgs/pull/91

chapulina, 2020-12-21:
Can this issue be closed?

JShep1, 2020-12-28:
Yep, closing as support now exists for adding model dependencies. Relevant PRs listed above.

closed 2020-12-28 by JShep1