gazebosim/gz-fuel-tools #113: Command line tools failing on Windows and macOS

chapulina, 2020-08-26:
The `ign_TEST`s fail on homebrew with:

```
31: Expected: (output.find(g_version)) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
31: I cannot find any available 'ign' command:
31: 	* Did you install any ignition library?
31: 	* Did you set the IGN_CONFIG_PATH environment variable?
31: 	    E.g.: export IGN_CONFIG_PATH=$HOME/local/share/ignition
```

And on Windows with:

```
31: D:\Jenkins\workspace\ignition_fuel-tools-ci-pr_any-windows7-amd64\ws\ign-fuel-tools\src\ign_TEST.cc(57): error: Expected: (output.find(g_version)) != (std::string::npos), actual: 18446744073709551615 vs 18446744073709551615
31: 'ign' is not recognized as an internal or external command,
31: operable program or batch file.
```

cheng-kevin, 2020-09-09:
Hi, I am seeing the same error in Ubuntu 18.04.5 LTS.

simonbogh, 2020-09-24:
Experiencing the same issue on macOS Catalina 10.15.6. Did export the path but I still receive the same error and gazebo will not launch.

chapulina, 2020-09-24:
> gazebo will not launch

Related issue: https://github.com/ignitionrobotics/ign-gazebo/issues/25

chapulina, 2021-01-12:
Possible fix coming from https://github.com/ignitionrobotics/ign-tools/issues/7