gazebosim/gz-fuel-tools #117: CGMClassic model files downloaded via ign-fuel-tools API do not match the model files at https://app.ignitionrobotics.org/fuel/models

ojasjoshi, 2020-08-27:
A bad reference to texture file(```Carts_Wheels_Diffuse.png```) in ```CGMClassic.mtl``` seems to have been corrected in the latest model file available at ```https://app.ignitionrobotics.org/OpenRobotics/fuel/models/CGMClassic```.

However, when pulled from ```ign-fuel-tools``` API, the older version of the file is downloaded. I have not verified the versions of other model files but the API might be referring to stale model versions.

chapulina, 2020-08-27:
How are you pulling the file? Did you check if the URL you're using is asking for a specific version? Did you delete the old model from the fuel cache in `~/.ignition/fuel`?

ojasjoshi, 2020-08-28:
Yes, fuel cache was cleared from ~/.ignition/fuel. 
 
I am pulling via [the download script in the example directory](https://github.com/ignitionrobotics/ign-fuel-tools/blob/master/example/download.cc) via recommended usage:
```bash
./download -t model -o OpenRobotics -n CGMClassic
```

We have tried pulling directly from https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models as such [here](https://github.com/aws-robotics/aws-robomaker-hospital-world/blob/577dc308382334dbf3c55d5a7d3e9ca295d26585/fuel_utility.py#L27). However, this gives a CGMClassic model version having a bad reference to the texture file(Carts_Wheels_Diffuse.png) in CGMClassic.mtl.

The error message we get following this is,
```bash
[Wrn] [FuelModelDatabase.cc:248] URI not supported by Fuel [Carts_Wheels_Diffuse]
[Wrn] [SystemPaths.cc:464] File or path does not exist [""] [Carts_Wheels_Diffuse]
``` 

This has however been corrected in the model version file located at https://app.ignitionrobotics.org/OpenRobotics/fuel/models/CGMClassic.

What is the ignition recommended URL/ method to pull the [latest versions of fuel models](https://app.ignitionrobotics.org/OpenRobotics/fuel/models/CGMClassic)?

ojasjoshi, 2020-08-28:
I see that ```https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/CGMClassic/2/CGMClassic.zip``` seems to have reference to the latest model version. 

If possible, can you please point me to resources stating how these urls are updated with new releases? Also, do you recommend updating these URLs in our code manually or is there a ignition preferred way of pulling the latest fuel models?

chapulina, 2020-08-28:
> What is the ignition recommended URL/ method to pull the latest versions of fuel models?

I recommend you use the `ign fuel download` command line tool like explained here: https://ignitionrobotics.org/api/fuel_tools/4.0/cmdline.html

i.e.

`ign fuel download -u https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/CGMClassic -v 4`

> can you please point me to resources stating how these urls are updated with new releases?

Each model is uploaded separately by the model owner. They're not tied to releases. There's no fixed schedule to update the models in the `OpenRobotics` org, they're updated as needed.

chapulina, 2020-08-28:
I'll close this for now since there doesn't seem to be a bug on `ign-fuel-tools`. If you have more usage questions, feel free to post them at https://answers.gazebosim.org/ :v:

closed 2020-08-28 by chapulina