gazebosim/gz-fuel-tools #118: Missing .mtl files for collision meshes in some fuel models

ojasjoshi, 2020-08-27:
```OpScrubs```, ```PatientWheelChair```, ```VisitorChair``` models do not have ```.mtl``` files for their collision meshes. This is resulting in a error message suggesting default materials to be used for these models. 

Is this intended from ignition or are these files missing by error?

chapulina, 2020-08-27:
I see an `OpScrubs_Col.mtl` here:

https://app.ignitionrobotics.org/OpenRobotics/fuel/models/OpScrubs

Could you be more specific about the error you're seeing and how you are getting the model?

chapulina, 2020-08-27:
I should also point out that this is probably not the correct place to post this issue. This reository is for the C++ client library and command line tool.

For issues with the contents of models, please refer to the Information link on the left here: https://app.ignitionrobotics.org/dashboard

ojasjoshi, 2020-08-28:
Correction: ```VisitorChair``` model does not have a .mtl file for the collision mesh.  

Error:
```bash
[Err] [OBJLoader.cc:79] WARN: Material file [fuel_models/VisitorChair/meshes/VisitorChair_Col.mtl ] not found. Created a default material.
[Wrn] [OBJLoader.cc:141] Missing material for shape[VisitorChair_Col_Cube.009] in OBJ file[/home/ANT.AMAZON.COM/ojasjosh/git_repos/aws-robomaker-hospital-world/fuel_models/VisitorChair/meshes/VisitorChair_Col.obj]
```

Sorry about creating the issue here. Would you recommend creating a support ticket [here](https://openrobotics.freshdesk.com/support/home)?

ojasjoshi, 2020-08-28:
Correction: ```VisitorChair``` model does not have a .mtl file for the collision mesh.  

Error:
```bash
[Err] [OBJLoader.cc:79] WARN: Material file [fuel_models/VisitorChair/meshes/VisitorChair_Col.mtl ] not found. Created a default material.
[Wrn] [OBJLoader.cc:141] Missing material for shape[VisitorChair_Col_Cube.009] in OBJ file[fuel_models/VisitorChair/meshes/VisitorChair_Col.obj]
```

Sorry about creating the issue here. Would you recommend creating a support ticket [here](https://openrobotics.freshdesk.com/support/home)?

chapulina, 2020-08-28:
> Would you recommend creating a support ticket here?

Yes, I think that's the best place right now.

> Sorry about creating the issue here. 

No worries, I'll close it now :wink:

closed 2020-08-28 by chapulina