gazebosim/gz-fuel-tools #119: Installation Issue

cheng-kevin, 2020-09-09:
#### Environment:
Ubuntu 18.04.5 LTS
Melodic/Gazebo9

#### Precondition:
-Built and installed ign-tools from source by following tutorial
-Installed libignition-fuel-tools-dev deb by following tutorial
-Both Installation completed with no errors.

Running the command:
`ign fuel list --help`

Returns:
```The 'ign' command provides a command line interface to the ignition tools.

  ign <command> [options]

List of available commands:

  help:          Print this help text.
  topic:         Print information about topics.
...
```

It seems that ign-fuel-tools was not properly installed.

cheng-kevin, 2020-09-09:
I also tried to install ign-fuel-tools from source by following the tutorial, but got stuck when resolving cmake errors.  The required dependency "ignition-msgs6" does not seem to be available (libignition-msgs1 through libignition-msgs5 were all available)

chapulina, 2020-09-10:
It looks like you also have `ign-transport` installed, how did you install it? 

Also, do you know the exact version of `ign-fuel-tools` that you're interested in? If you're having problems with `ign-msgs6`, you're probably trying to install the master branch, which hasn't been release yet, and that may not be what you want.

cheng-kevin, 2020-09-10:
I believe I have installed ignition-transport via apt-get (not via source)

Thanks for note regarding the master branch, I have since pulled the ign-fuel-tools4 branch, built, and installed.   I'm not sure the exact version I need, but any version compatible with my environment (listed above) will suffice.  However running
`ign fuel list --help`

still produces the same result.

Is there any documentation about what versions of ign-tools is compatible with which version of ign-fuel-tools/gazebo/various required ignition packages.

Heres a list of whats installed/
'''
libignition-cmake-dev/unknown,now 0.6.1-1~bionic amd64 [installed]
libignition-cmake2-dev/unknown,now 2.5.0-1~bionic amd64 [installed]
libignition-common/unknown,now 1.1.1-1~bionic amd64 [installed,automatic]
libignition-common-dev/unknown,now 1.1.1-1~bionic amd64 [installed]
libignition-common3/unknown,now 3.6.0-1~bionic amd64 [installed]
libignition-common3-av/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-av-dev/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-core-dev/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-dev/unknown,now 3.6.0-1~bionic amd64 [installed]
libignition-common3-events/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-events-dev/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-graphics/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-graphics-dev/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-profiler/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-common3-profiler-dev/unknown,now 3.6.0-1~bionic amd64 [installed,automatic]
libignition-fuel-tools1-1/unknown,now 1.2.0-1~bionic amd64 [installed,auto-removable]
libignition-fuel-tools1-dev/unknown,now 1.2.0-1~bionic amd64 [installed,auto-removable]
libignition-math4/bionic,now 4.0.0+dfsg1-4 amd64 [installed,automatic]
libignition-math4-dev/bionic,now 4.0.0+dfsg1-4 amd64 [installed,automatic]
libignition-math6/unknown,now 6.5.0-2~bionic amd64 [installed,automatic]
libignition-math6-dev/unknown,now 6.5.0-2~bionic amd64 [installed,automatic]
libignition-msgs/bionic,now 1.0.0+dfsg1-5 amd64 [installed,automatic]
libignition-msgs-dev/bionic,now 1.0.0+dfsg1-5 amd64 [installed]
libignition-msgs5/unknown,now 5.3.0-1~bionic amd64 [installed,automatic]
libignition-msgs5-dev/unknown,now 5.3.0-1~bionic amd64 [installed]
libignition-transport4/bionic,now 4.0.0+dfsg-4 amd64 [installed,automatic]
libignition-transport4-dev/bionic,now 4.0.0+dfsg-4 amd64 [installed]
'''

chapulina, 2020-09-10:
> any version compatible with my environment (listed above) will suffice.

For Gazebo 9, it's `ign-fuel-tools1`. We don't have a good place describing this besides the [source code](https://github.com/osrf/gazebo/blob/2f3024dc7d76159573090e312d1f6839d09d5c2f/cmake/SearchForStuff.cmake#L741).

> Is there any documentation about what versions of ign-tools is compatible with which version of ign-fuel-tools/gazebo/various required ignition packages.

For Gazebo <= 10 we don't, but Gazebo 11 uses the Citadel collection and you can see the versions listed here: https://ignitionrobotics.org/docs/citadel

> Heres a list of whats installed

I don't see `ignition-tools`, which probably means you're using that library from source. If that's the case, I recommend uninstalling the library from source and installing from debs.

---

I also checked the files installed by `libignition-fuel-tools1*` and it looks like they're [not installing](https://github.com/ignition-release/ign-fuel-tools-release/blob/master/ubuntu/debian/libignition-fuel-tools.install) the file that's needed for the command line tool to work, [fuel.yaml](https://github.com/ignitionrobotics/ign-fuel-tools/blob/ign-fuel-tools1/conf/fuel.yaml.in). This looks like a mistake to me, because the package depends on `libignition-tools-dev` and installs other files related to the command line, like `/usr/lib/ruby/ignition/cmdfuel1.rb`. 

This means that other Fuel functionality should be working, even though the command line isn't.

It also means that you shouldn't have the same problem using newer versions like 3 and 4. You can try installing that to use the command line tools: `libignition-fuel-tools4*`. I believe the models downloaded using that version should work with Gazebo 9.

cheng-kevin, 2020-09-10:
Thank you @chapulina ! Following your recommendation has been successful.

closed 2020-09-10 by cheng-kevin