gazebosim/gz-fuel-tools #150: Add option to skip cached models when downloading from command line

chapulina, 2020-12-23:
Downloading a collection can be time consuming *, so the user may want to do it in chunks. But every new call to download a collection triggers a re-download of all its models. It would be nice to have a flag, like `--skip-cached` to say they should be skipped.

For example, I could start normally:

`ign fuel download -j 4 -v 4 -u "https://fuel.ignitionrobotics.org/1.0/openrobotics/collections/Gazebo classic model database"`

Then cancel halfway with `Ctrl + C`

Then restart with:

`ign fuel download -j 4 -v 4 --skip-cached -u "https://fuel.ignitionrobotics.org/1.0/openrobotics/collections/Gazebo classic model database"`

> \* or,  in my case, `ign-fuel-tools` crashes halfway - but that's another problem

khaledgabr77, 2025-04-10:
I’ll take care of this issue. Please assign it to me.

kscottz, 2025-04-14:
@khaledgabr77 how is it going?

khaledgabr77, 2025-04-15:
@kscottz It's in progress.

khaledgabr77, 2025-04-23:
@kscottz 
Last week was very busy, so I couldn’t spend much time on this. I added the `--skip-cached` option to the Gazebo Fuel tools and implemented it, but it is not working yet. I will try again this week.
Thank you.

khaledgabr77, 2025-04-24:
Hi @kscottz 
I tested the `--skip-cached` flag, and it's working fine now. As shown in the image, it correctly skips models that were already downloaded. I’ll create the PR tomorrow.

- The first image without `--skip-cached`:

[image]

- The second image with `--skip-cached`:

[image]

khaledgabr77, 2025-04-25:
@kscottz  You can find the pull request at PR #461.

kscottz, 2025-04-28:
@khaledgabr77 you are a champ! Thanks so much, I'll take a look!