gazebosim/gz-fuel-tools #168: The `--owner` flag in `ign fuel list` switches the command to search only cached models

peci1, 2021-03-02:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? Binary 5.1.1

## Description
* Expected behavior: `ign fuel list -o openrobotics -t model` should list all models from openrobotics available on Fuel.
* Actual behavior: It only lists locally cached models when run with a non-empty cache.

## Steps to reproduce

1. `mkdir /tmp/fuel_test1`
2. `IGN_FUEL_CACHE_PATH=/tmp/fuel_test1 ign fuel list -o openrobotics --type model | wc -l`

   815

3. `mkdir /tmp/fuel_test2`
4. `IGN_FUEL_CACHE_PATH=/tmp/fuel_test2 ign fuel download -u "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 4 Lights"`
5. `IGN_FUEL_CACHE_PATH=/tmp/fuel_test2 ign fuel list -o openrobotics --type model | wc -l`

   8

This is at least unexpected. I'd expect either a switch that would allow me choosing between all and cached models, or a warning/info message, that would tell me that the command uses only what it found in the local cache.

When called without the `--owner` flag, the command seems to always contact the fuel database and downloads the list of all models.

chapulina, 2021-03-02:
I agree that it's not a behaviour the user would expect. Here's where the cache is checked:

https://github.com/ignitionrobotics/ign-fuel-tools/blob/4bb5bfa8eabad7f11669d6296f6b59ff921427ff/src/FuelClient.cc#L383-L388

At a minimum, I think we could offer a flag to only check the cache, or check the server.

peci1, 2021-03-03:
I think a CLI flag would be the best. I just can't decide what should be the default behavior. But I'm a little bit more inclined for searching the online DB by default.

chapulina, 2021-03-03:
As a user, I'd expect to get all of an owner's models with that invocation, online or otherwise. So I agree that searching online would be a good default behaviour.