gazebosim/gz-fuel-tools #172: Fuel model error: https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 6 Rails

peci1, 2021-03-16:
I'm not sure if this is the best place to report broken models, but I haven't found any better. There is a report button on app.ignitionrobotics.org, and it lets me select the reason "Broken model", but it won't let me enter any details.

Model https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Tunnel%20Tile%206%20Rails has broken metadata.

The metadata.pbtxt contains the following part:

```
dependencies {
  uri: "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 6"
  uri: "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Rail Tunnel Elevation"
}
```

And `ign fuel download` looks like:

```
$ ign fuel download -v4 --url "https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Tunnel Tile 6 Rails/0"
Downloading model: 
  Name: tunnel tile 6 rails
  Owner: openrobotics
  Server:
    URL: https://fuel.ignitionrobotics.org
    Version: 1.0

[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/metadata.pbtxt]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/model.config]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/model.sdf]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/1.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/2.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/3.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/4.jpg]
[Dbg] [Zip.cc:175] Created file [/home/peci1/.ignition/fuel/fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails/1/thumbnails/5.jpg]
[libprotobuf ERROR google/protobuf/text_format.cc:298] Error parsing text-format ignition.msgs.FuelMetadata: 23:6: Non-repeated field "uri" is specified multiple times.
Download succeeded.
```

Looking at [the definition of the metadata message](https://github.com/ignitionrobotics/ign-msgs/blob/ign-msgs6/proto/ignition/msgs/fuel_metadata.proto), it seems to me the metadata might really be wrong as it contains only one `dependencies` section with two `url` fields.

There's also comment

https://github.com/ignitionrobotics/ign-msgs/blob/521b861bc7cfe8276a38ad58409f1b6307a88e17/proto/ignition/msgs/fuel_metadata.proto#L144

which is obviously out of date now.

chapulina, 2021-03-16:
Since that model is part of the SubT collection, I think https://github.com/osrf/subt is a better place to report this.

peci1, 2021-03-16:
Ok, transferred there. https://github.com/osrf/subt/issues/824 .

closed 2021-03-16 by peci1