gazebosim/gz-fuel-tools #174: Gateway timeout (504) or Service unavailable (503) on fuel.ignitionrobotics.org

peci1, 2021-03-17:
## Environment
* OS Version: Ubuntu 18.04
* Source or binary build? Binary 5

## Description
* Expected behavior: ign fuel download works
* Actual behavior: 

```
$ ign fuel download -v 4 -j 8 --type model -u "https://fuel.ignitionrobotics.org/OpenRobotics/collections/SubT Tech Repo"
Downloading collection: 
  Name: SubT Tech Repo
  Owner: OpenRobotics
  Server:
    URL: https://fuel.ignitionrobotics.org
    Version: 1.0

[Dbg] [ModelIter.cc:199] Got response [{"errcode":3017,"msg":"Page not found","extra":null,"errid":"4e518156-fd5c-4621-80c5-2f9e271c7eec","route":"GET /1.0/OpenRobotics%2Fcollections%2FSubT%20Tech%20Repo%2Fmodels?page=18","user-agent":"FuelTools 5.1.1","remote-address":"147.32.84.59"}
]
[Msg] Found 322 models in collection [SubT Tech Repo]
[Msg] Using 8 jobs to download collection of 322 items
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 6 rails]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 5 rails]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 4 rails]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 2 rails]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 4 lights]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 1 lights]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 2 lights]
[Msg] Downloading model [https://fuel.ignitionrobotics.org/openrobotics/models/tunnel tile 3 lights]
[Err] [FuelClient.cc:582] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/tunnel tile 4 rails/tip/tunnel tile 4 rails.zip
  REST response code: 504
[Err] [FuelClient.cc:582] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/tunnel tile 6 rails/tip/tunnel tile 6 rails.zip
  REST response code: 504
[Err] [FuelClient.cc:582] Failed to download model.
  Server: https://fuel.ignitionrobotics.org
  Route: openrobotics/models/tunnel tile 5 rails/tip/tunnel tile 5 rails.zip
  REST response code: 504
```

and so on.

It's been hapenning for a few hours now.

I also saw this:

```
$ ign fuel download -v 4 -j 8 --type model -u "https://fuel.ignitionrobotics.org/OpenRobotics/collections/SubT Tech Repo"
Downloading collection: 
  Name: SubT Tech Repo
  Owner: OpenRobotics
  Server:
    URL: https://fuel.ignitionrobotics.org
    Version: 1.0

[Msg] Found 0 models in collection [SubT Tech Repo]
There are no items in collection [SubT Tech Repo]
```

chapulina, 2021-03-17:
We're on it, updates will be posted here.

iche033, 2021-03-17:
we've upgraded Fuel's database instance. It seems to be working fine now but will keep an eye on it.

peci1, 2021-03-17:
Thanks. I was able to download whole SubT collection shortly after chapulina's comment.

closed 2021-03-17 by peci1