gazebosim/gz-fuel-tools #186: Upload a model from the command line to an organization

diegoferigo, 2021-05-12:
I couldn't find a way to upload a model using the `ign fuel` command line to an organization instead of the personal profile.

The [command line tutorial](https://github.com/ignitionrobotics/ign-fuel-tools/blob/ign-fuel-tools6/tutorials/03_command_line.md) does not cover resource upload, and [the example in the main README](https://github.com/ignitionrobotics/ign-fuel-tools#examples) does not show the possibility to upload the resource to an organization.

## Desired behavior

Being able to upload a model to an organization instead of the personal profile.

## Implementation suggestion

If it is not currently possible, maybe a new `--owner|-o` option could be added?

```bash
ign fuel upload \
    -m ~/path/to/model/folder \
    --owner <organization> \  # New option?
    -u https://fuel.ignitionrobotics.org/ \
    --header 'Private-Token: <token>'
```

## Alternatives considered

It is possible to do it from the web app, where a drop-down menu is available in the _Owner_ field.

chapulina, 2021-05-12:
This was recently added in https://github.com/ignitionrobotics/ign-fuel-tools/pull/179 to Dome.

> The command line tutorial does not cover resource upload

It would be good to update the docs.

diegoferigo, 2021-05-13:
Ow ok, thanks @chapulina, I missed that PR. I'm currently on Edifice with a colcon setup, and that PR hasn't been forward ported yet. I inspected the code and found no related logic, without thinking to check better the previous PRs. Feel free to close this issue if you don't want to use it as a reminder for the documentation.

chapulina, 2021-05-13:
Gotcha. We'll be forward-porting that soon.

I updated this issue to be about the missing documentation.

Theonewhomadethings, 2025-04-05:
Hi @chapulina 

I'd like to take on this issue — I saw it’s marked as a good first issue and seems like a great way to get started.

Since the last update here was in 2021, are there any version compatibility changes I should be aware of before proceeding?

Could you please assign this to me?

Thanks!

ahcorde, 2025-04-07:
HI @Theonewhomadethings 

I assigned the issue to you, This library doesn't not change that much during the last years

Theonewhomadethings, 2025-04-12:
Hi everyone @ahcorde ,

I need a little bit of advice on the upload model from the command line for this tool.

I’m working on this documentation issue and trying to verify the command‐line upload feature myself before I write it. However, I’m currently unable to upload a model to either my personal account or my organization. I keep getting HTTP errors (400 or 500). Here’s a quick overview of what I’ve tried:

Setup:

Created a new Fuel account and verified my email.

Generated a fresh access token under Settings → Access Tokens.

(Optional) Created an organization for testing --owner <organization_name>.

CLI Environments:

Binary install from packages.osrfoundation.org (Ubuntu 22.04).

Source install of gz-fuel-tools and gz-tools (hoping to ensure the --owner feature was present).

In both cases, gz fuel can download models just fine, but upload fails with codes 400 or 500.

Sample Commands (for personal account):

gz fuel upload \
  -m ~/test_box_model \
  -u https://fuel.gazebosim.org \
  --header "Private-Token: <MY_TOKEN>" \
  -v 4

The model.config includes a <sdf version="1.6">model.sdf</sdf> tag, so parsing that part seems okay now.

Errors:

Sometimes REST response code: 400 (Bad Request) on my test model I made

Other times REST response code: 500 on a downloaded model that already exists from a random user that I am re uploading as a test to my personal account or org.

I tried renaming the model to avoid collisions or special characters. Also tried the --owner flag with my exact username. No luck so far.

Request:

Could someone with a working setup share a successful example command snippet (with your token redacted) that shows how you upload a model to your personal account?

Also, if you see any differences in how you structure your model.config / model.sdf, I’d love to know.

I’m happy to update the documentation once I figure out the right approach. Any guidance would be really appreciated—thank you!

kscottz, 2025-04-15:
@ahcorde could you offer some advice here please. 💟

kscottz, 2025-04-21:
@ahcorde  bump