gazebosim/gz-fuel-tools #189: New release for ign-fuel-tools4/Citadel?

traversaro, 2021-06-11:
## Desired behavior
Hi everyone! For Gazebo 11 Windows packaging, it would be nice to make a release of the ign-fuel-tools4 branch, as it includes a critical bug fix for Windows: https://github.com/ignitionrobotics/ign-fuel-tools/pull/178 . The required changes in ignition-common3 (https://github.com/ignitionrobotics/ign-common/pull/197) were already release in 3.13 (see https://github.com/ignitionrobotics/ign-common/tags).

cc @TobiasFischer

## Alternatives considered
An alternative we tried in https://github.com/conda-forge/libignition-fuel-tools-feedstock/pull/32#issuecomment-859291664 was to backport the patch to an earlier release, but unfortunately it does not apply cleanly. 

## Implementation suggestion
I guess following the standard release process will be ok.

## Additional context
This came out in https://github.com/conda-forge/libignition-fuel-tools-feedstock/pull/32#issuecomment-859291664 .

Tobias-Fischer, 2021-06-21:
Seems like this can be closed - thanks @chapulina!

closed 2021-06-21 by ahcorde

traversaro, 2021-06-21:
Thanks a lot!