gazebosim/gz-fuel-tools #198: Add hash/checksum/md5 to object info in REST API

FirefoxMetzger, 2021-08-05:
## Desired behavior

```
"https://fuel.ignitionrobotics.org/1.0/{username}/models/{model_name}"
```

The endpoint above provides metadata about the model at this endpoint, which is very cool. An example response may look like this:

```
Query: https://fuel.ignitionrobotics.org/1.0/OpenRobotics/models/Construction%20Cone
```

```json
{
  "createdAt": "2018-01-27T00:22:54Z",
  "updatedAt": "2021-08-05T16:56:43Z",
  "name": "Construction Cone",
  "owner": "OpenRobotics",
  "description": "An orange construction cone",
  "likes": 0,
  "downloads": 3710,
  "filesize": 622427,
  "upload_date": "2018-01-27T00:22:53Z",
  "modify_date": "2018-01-27T00:22:53Z",
  "license_id": 1,
  "license_name": "Creative Commons Zero v1.0 Universal",
  "license_url": "https://creativecommons.org/publicdomain/zero/1.0/",
  "license_image": "https://i.creativecommons.org/p/88x31.png",
  "permission": 0,
  "url_name": "",
  "thumbnail_url": "/OpenRobotics/models/Construction%20Cone/tip/files/thumbnails/1.png",
  "version": 2,
  "private": false
}
```

What would be very cool is to have it include some form of checksum (e.g. md5). This way a client can verify that nothing nefarious or unfortunate happened while downloading the model and that it is indeed the correct model.

My background is that I am currently writing unit tests for the fuel-support that I added to ropy's SDF bindings (see this forum discussion for a brief primer on the bindings: https://community.gazebosim.org/t/sdformat-data-bindings-for-python-all-sdf-versions/1054/4).

Currently, I download the file externally, compute the md5, and then download it again with ropy and compute that md5 to see if (a) it crashes, and (b) if it downloads the correct file. It would be nice (for server load and test speed) if this could be done with one call to the model info (much lighter) and one single download of the full model (by ropy).

chapulina, 2021-08-05:
I'll just drop a link here to the web server's source code, since this issue would probably be tackled there:

https://gitlab.com/ignitionrobotics/web/fuelserver

FirefoxMetzger, 2021-08-05:
Thanks for the pointer @chapulina . I migrated the issue there. Should this one be closed in favor of the other?

Link: https://gitlab.com/ignitionrobotics/web/fuelserver/-/issues/9

chapulina, 2021-08-05:
> Should this one be closed in favor of the other?

Yeah let's do that. Thanks!

closed 2021-08-05 by chapulina