gazebosim/gz-fuel-tools #202: stderr ignition-fuel_tools6 while building

ggiulian, 2021-08-28:
## Environment
* OS Version:  20.04 
* Source or binary build?
source: ROS2 Galactic and Ignition Edifice

    
<details>
  When building there is this stderr referred to developers and authors of ignition fuel tools
```
# paste log here
```stderr: ignition-fuel_tools6             
CMake Warning (dev) at /home/ggiuliano/edifice_ws/install/share/cmake/ignition-cmake2/cmake2/IgnUtils.cmake:385 (message):
   -- THIS MESSAGE IS INTENDED FOR IGNITION-FUEL_TOOLS AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for YAML.  It was not provided by the
  find-module for the package, nor was it explicitly passed into the call to
  ign_find_package(~).  This is most likely an error in this project's use of
  ign-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:42 (ign_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< ignition-fuel_tools6 [6min 10s]

</details>

## Description
* Expected behavior: No stderr while building
* Actual behavior: sderr while building

chapulina, 2021-08-30:
Do you have libyaml installed? What version is it?

ggiulian, 2021-08-30:
Yes, the version is the newest: 0.2.2-1

chapulina, 2021-08-30:
Interesting, I have the same version and it works for me, and it also works on CI.

We'll need more information about what steps you took to get to this problem.

ggiulian, 2021-08-31:
Mainly, I followed this documentation: https://ignitionrobotics.org/docs/edifice/install_ubuntu_src.
leaving out the installations of generic tools, vcs and colcon from pip or apt (because already installed) and git. So I created the workspace on the home, after that I followed the instructions of the documentation. I have left out the installation of gcc and g ++, since I work on Focal. Eventually I built everything with `colcon build --merge-install`

ggiulian, 2021-08-31:
I tried with different build like `colcon build --cmake-args -DBUILD_TESTING=OFF --merge-install` and `colcon build --packages-select PACKAGE_NAME` but i had the same stderr

bperseghetti, 2024-06-18:
@azeey @j-rivero appears this might still be a lingering issue:
```
--- stderr: gz-fuel_tools9                                                                                                                                                  
CMake Warning (dev) at /home/user/git/harmonic/install/share/cmake/gz-cmake3/cmake3/GzFindPackage.cmake:456 (message):
   -- THIS MESSAGE IS INTENDED FOR GZ-FUEL_TOOLS AUTHORS --
      (IF YOU SEE THIS, PLEASE REPORT IT)

  Could not find pkg-config information for YAML.  It was not provided by the
  find-module for the package, nor was it explicitly passed into the call to
  gz_find_package(~).  This is most likely an error in this project's use of
  gz-cmake.
Call Stack (most recent call first):
  CMakeLists.txt:44 (gz_find_package)
This warning is for project developers.  Use -Wno-dev to suppress it.

---
Finished <<< gz-fuel_tools9 [26.1s]
```

azeey, 2024-06-18:
I was able to reproduce this if I source a ROS humble installation with `ros-humble-libyaml-vendor` installed. I suspect this is fixed in Jazzy since the `libyaml_vendor` package has been updated to use `ament_cmake_vendor_package`.