gazebosim/gz-fuel-tools #215: docs, ui: Possible to provide concrete link back to App? Possible to add `ign fuel` cmd to App?

EricCousineau-TRI, 2021-11-15:
## Desired behavior
With download example here: https://ignitionrobotics.org/libs/fuel_tools

Would be nice to:
- see the original model link from this, e.g. https://app.ignitionrobotics.org/OpenRobotics/fuel/models/Ambulance.
- see an example of downloading a specific version (e.g. model's version 1)

From web UI, it would be nice to see minimal `ign fuel download` command to retrieve model programmatically (incl. version by default?)

## Alternatives considered
N/A

## Implementation suggestion
Doc edit
(hopefully quick?) UI edit

## Additional context
Usage here: https://github.com/EricCousineau-TRI/repro/pull/73

azeey, 2021-12-16:
@EricCousineau-TRI, I've opened a ticket on the web app here: https://gitlab.com/ignitionrobotics/web/app/-/issues/15.

EricCousineau-TRI, 2021-12-17:
Thanks! Closing this in lieu of gitlab issue.

closed 2021-12-17 by EricCousineau-TRI