gazebosim/gz-fuel-tools #307: :farmer: 7 failing tests due to colon in `common::joinPaths`

Crola1702, 2022-10-27:
## Environment
* OS Version: 
  * Windows
* Source build
  * Fuel tools 8

## Description
* Expected behavior: Build passing successful
* Actual behavior: Build has 7 test regressions

## Steps to reproduce
1. Run a build in [ign_fuel-tools-gz-8-win](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/)
2. See 7 test regressions when you finish
3.

## Output

### Fuel tools 8 Windows | Test regressions | Pending investigation :yellow_circle: 

Reference build: https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/#showFailuresLink

Failing tests:
<details>

* [UNIT_gz_src_TEST.test_ran](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/UNIT_gz_src_TEST/test_ran/)
* [FuelClientTest.CachedWorld](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/FuelClientTest/CachedWorld/)
* [FuelClientTest.PatchModelFail](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/FuelClientTest/PatchModelFail/)
* [LocalCacheTest.MatchingModels](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/LocalCacheTest/MatchingModels/)
* [LocalCacheTest.MatchingModel](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/LocalCacheTest/MatchingModel/)
* [LocalCacheTest.AllWorlds](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/LocalCacheTest/AllWorlds/)
* [LocalCacheTest.MatchingWorlds](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/LocalCacheTest/MatchingWorlds/)
* [LocalCacheTest.MatchingWorld](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/19/testReport/junit/(root)/LocalCacheTest/MatchingWorld/)

</details>

---

FuelClientTest output:
```
C:\Jenkins\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient_TEST.cc:108
Value of: common::createDirectories(common::joinPaths(worldPath, "2"))
  Actual: false
Expected: true
```

LocalCacheTest output:
```
C:\Jenkins\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\LocalCache_TEST.cc:137
Value of: common::createDirectories(common::joinPaths(serverPath, "alice", "worlds", "am1", "2"))
  Actual: false
Expected: true
```

This test is probably failing because of [Windows doesn't support colons in filenames](https://github.com/gazebosim/gz-fuel-tools/issues/106) (reported 2 years ago).

Colon is used in this [line of code](https://github.com/gazebosim/gz-fuel-tools/blob/gz-fuel-tools8/src/FuelClient_TEST.cc#L106), and the test is failing 2 lines below.

azeey, 2023-01-09:
We need to replace the colon with something that's valid as a file name, but we're not sure what character.
Possible ideas:
- Replace with `.`
- base64 hash the host name.

Crola1702, 2023-01-26:
Any new ideas? @nkoenig

Crola1702, 2023-02-03:
> Any new ideas?

(Friendly ping @nkoenig)

azeey, 2023-04-10:
Let's disable this for now.

PR #351 (2023-04-25): Disable all LocalCache Tests for Windows
# 🦟 Bug fix

Fixes #307 (Partially)

## Summary
This PR disables LocalCache Tests for Windows

## Checklist
- [X] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test coverage](https://gazebosim.org/docs/all/contributing#test-coverage))
- [ ] While waiting for a review on you [... cut]
closed 2023-04-25 by azeey