gazebosim/gz-fuel-tools #315: :farmer: Flaky Download Model tests (`FuelClient`, `UNIT_gz_src_TEST`, `UNIT_Interface_TEST`)

Crola1702, 2022-11-21:
## Environment
* OS Version: Ubuntu Focal 20.04
* Source build?

## Description
* Expected behavior: `FuelClient`, `UNIT_gz_src_TEST`, and `UNIT_Interface_TEST` should not fail
* Actual behavior: These test sometimes fail when downloading the models

## Steps to reproduce

1. Run a build on https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/
2. See it fail if lucky.

## Output

I see [this PR](https://github.com/gazebosim/gz-fuel-tools/pull/310) made changes on [gz_src_TEST.cc](https://github.com/gazebosim/gz-fuel-tools/pull/310/files#diff-73cc6e9d1bcf5734cc886a3965135343197b62d2322c5c8332795d7756aa709bR195) for model downloading tests. 

#### Build [21](https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/21/consoleFull):
```
[Err] [FuelClient.cc:662] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: chapulina/models/test box/tip/test box.zip
  REST response code: 504
```

#### Build [20](https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/20/):
It passed (no errors dowloading any model)

#### Build [19](https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/19/consoleFull):

FuelClient:
```
[Err] [FuelClient.cc:662] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: chapulina/models/test box/tip/test box.zip
  REST response code: 504
```

#### Build [18](https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/18/consoleFull):

UNIT_Interface_TEST:
```
[Err] [FuelClient.cc:662] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openrobotics/models/bus/1/bus.zip
  REST response code: 504
```

UNIT_gz_src_TEST
```
[Err] [FuelClient.cc:662] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openroboticstest/models/backpack/tip/backpack.zip
  REST response code: 504
[Err] [FuelClient.cc:662] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openroboticstest/models/teambase/tip/teambase.zip
  REST response code: 504
```

nkoenig, 2022-11-22:
We need to update the Fuel server to use S3 as a download mechanism. Currently, a download can be affected by Fuel load.

nkoenig, 2022-11-22:
#316 will help

Crola1702, 2022-12-09:
We now have new [`gz_src_TEST.CollectionTest/DownloadCollectionTest`](https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/24/testReport/junit/(root)/CollectionTest_DownloadCollectionTest/) test regressions in gz-fuel-tools8 focal

Reference build: https://build.osrfoundation.org/job/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/24/

`CollectionTest/DownloadCollectionTest.AllItems` log output:
```
/home/jenkins/workspace/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/ign-fuel-tools/src/gz_src_TEST.cc:398
Value of: common::isDirectory( common::joinPaths(this->testCachePath, "fuel.gazebosim.org", "openroboticstest", "models", "backpack", "2"))
  Actual: false
Expected: true
```

`CollectionTest/DownloadCollectionTest.Models` log output:
```
/home/jenkins/workspace/ignition_fuel-tools-ci-gz-fuel-tools8-focal-amd64/ign-fuel-tools/src/gz_src_TEST.cc:461
Value of: common::isDirectory( common::joinPaths(this->testCachePath, "fuel.gazebosim.org", "openroboticstest", "models", "backpack", "2"))
  Actual: false
Expected: true
```

@nkoenig do you think this could be a `gz-common` or `fuel-tools` problem?

Crola1702, 2022-12-14:
Taking a close look at the issue, it seems the error is related to `common::joinPaths`. 

We already have an issue for that one, so I'll link it here: https://github.com/gazebosim/gz-fuel-tools/issues/307

It might be failing because: 
> This test is probably failing because of https://github.com/gazebosim/gz-fuel-tools/issues/106 (reported 2 years ago).

Crola1702, 2023-03-23:
Redirecting this issue to https://github.com/gazebosim/gz-fuel-tools/issues/340

PR #316 (2022-11-28): Reduce asset download sizes in Fuel_Client_TEST

# 🦟 Bug fix

This helps to fix #315 by using models that have been reduced in size. The changed models had their images and meshes removed, and move to the `openroboticstest` account on Fuel.

## Checklist
- [x] Signed all commits for DCO
- [ ] Added tests
- [ ] Updated documentation (as needed)
- [ ] Updated migration guide (as needed)
- [ ] Consider updating Python bindings (if the library has them)
- [ ] `codecheck` passed (See [contributing](https://gazebosim.org/docs/all/contributing#contributing-code))
- [ ] All tests passed (See [test cov [... cut]
closed 2023-03-23 by Crola1702