gazebosim/gz-fuel-tools #349: gz fuel download returns 0, even when the download failed

dHofmeister, 2023-04-04:
## Environment
* OS Version: Ubuntu 22.04, binary build
* gz-fuel-tools:
    type: git
    url: https://github.com/gazebosim/gz-fuel-tools
    version: gz-fuel-tools8
    commit: 54491c8

## Description
* Expected behavior: return code non-zero
* Actual behavior:error code zero

```
root@user-desktop:/gazebo# gz fuel download --url https://fuel.gazebosim.org/1.0/openrobotics/models/sydney_regatta
[Err] [FuelClient.cc:677] Failed to download model.
  Server: https://fuel.gazebosim.org
  Route: openrobotics/models/sydney_regatta/tip/sydney_regatta.zip
  REST response code: 504
Download failed.
root@user-desktop:/gazebo# echo $?
0
```

It would be nice to have a non-zero return code in the terminal, as it would make error handling and catching easier. 

Greetings

M1chaelM, 2023-06-25:
This bug is particularly bad when working with Docker images. Since resources are not typically downloaded and cached until the first time they are needed, users running `gz sim` commands from a Docker image have to download everything from fuel every time unless they use `gz fuel download` to build the cache into the image. However, Docker relies on the error code to determine whether a step in its build process has succeeded. So when `gz fuel download` returns 0 even though it has failed, the result is that Docker does not detect the error, and bakes the failed download into the image. 

If the user is not actively reading the console output they will think they've cached the resource when they haven't. Further, to fix the problem, the user will have to manually add a cache-bust argument to the appropriate spot in the Dockerfile, since Docker has no way of knowing which layer to start from.

Rahul-K-A, 2025-04-02:
I can make a quick fix for this, if we can decide on the return values. Looking at the latest commit in gz-fuel-tools8: https://github.com/gazebosim/gz-fuel-tools/blob/e7ac6d403a694b3817fa324324ccc9424dd1b00a/example/download.cc#L112-L114, i see that `download` returns a `-1` on failure and a `0` on success. Same thing holds true for gz-fuel-tools10 Does `0` on failure and `1` on success work instead? Since gz-fuel-tools8 doesnt seem to be in active development, shall I push the changes directly in gz-fueld-tools10 instead?

Rahul-K-A, 2025-04-02:
Looking into the source code. it seems the only change that I need to make is replacing
```
if (client.DownloadWorld(id))
    return 0
```
to

```
 if (client.DownloadWorld(id).Type() == ResultType::FETCH)
   return 0
```

Could someone confirm?

kscottz, 2025-04-02:
@Rahul-K-A That seems legitimate to me. Send over a PR and we can give it a try.

Rahul-K-A, 2025-04-03:
@kscottz just to confirm (since this is my first time contributing to gazebo), I make a branch off of the `main` branch, correct?

Rahul-K-A, 2025-04-03:
PR is open: https://github.com/gazebosim/gz-fuel-tools/pull/459 . Please let me know if there's anything that I need to change.

Rahul-K-A, 2025-04-03:
Also, I am not sure if this comes under the purview of this specific ticket, but I noticed that we return `-1` on failure. Some operating systems use an unsigned 8-bit integer to store the return values, so returning `-1` might cause an underflow and make the shell return `255` when doing `echo $?`. Would returning `0` for failure and `1` be more compatible?

Rahul-K-A, 2025-04-03:
Additionally, in `gz.cc`, when downloading worlds and models in a collection, there doesnt seem to be any specific error handling. We get back a vector of results, but it doesnt look like we are parsing it and displaying warnings for missing models or worlds. I am not attempting to fix this as part of the PR but I think adding in a result parser for collections might be useful in the future.

https://github.com/gazebosim/gz-fuel-tools/blob/e7ac6d403a694b3817fa324324ccc9424dd1b00a/src/gz.cc#L626-L634

azeey, 2026-09-24:
The issue has be fixed by #459. Please open a separate issue for any remaining bugs

closed 2026-09-24 by azeey