gazebosim/gz-fuel-tools #373: :farmer: FuelClientTest CachedWorld and PatchModelFail failing

Crola1702, 2023-09-07:
## Environment
* OS Version: Windows
* Source gz-fuel-tools 

## Description
* Expected behavior: Test should pass
* Actual behavior: Test is failing due to a common::createDirectories failure

## Steps to reproduce

1. Run a build in gz-fuel-tools windows 8, 9 or ci-main
2. See FuelClientTest fail

## Output

Reference build: https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/73/

Test regressions:
* [FuelClientTest.CachedWorld](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/73/testReport/junit/(root)/FuelClientTest/CachedWorld/)
* [FuelClientTest.PatchModelFail](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/73/testReport/junit/(root)/FuelClientTest/PatchModelFail/)

Both PatchModelFail and CachedWorld have:

```
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\Filesystem.cc:50] Failed common::createDirectories (ec: system:267 The directory name is invalid.)
```

<summary>
CachedWorld Log output:
</summary>

```
5: [ RUN      ] FuelClientTest.CachedWorld
5: [Dbg] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient_TEST.cc:95] Creating local world in [C:\Windows\TEMP\gz\testa05104]
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\Filesystem.cc:50] Failed common::createDirectories (ec: system:267 The directory name is invalid.)
5: C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient_TEST.cc(100): error: Value of: common::createDirectories(common::joinPaths(worldPath, "2"))
5:   Actual: false
5: Expected: true
```

</details>

<details>
<summary>
PatchModelFail Log output:
</summary>

```
5: [ RUN      ] FuelClientTest.PatchModelFail
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\URI.cc:451] Adding empty path segment to URI  has no effect.
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\URI.cc:451] Adding empty path segment to URI models has no effect.
5: [Err] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient.cc:1684] The model path[bad] doesn't exist.
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\URI.cc:451] Adding empty path segment to URI  has no effect.
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\URI.cc:451] Adding empty path segment to URI models has no effect.
5: [Err] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient.cc:1724] Provided model directory[C:\Windows\TEMP\gz\testa05104] needs a metadata.pbtxt or a model.confg file. [Dbg] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient_TEST.cc:40] Creating local model in [C:\Windows\TEMP\gz\testa05104]
5: [Wrn] [C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-common\src\Filesystem.cc:50] Failed common::createDirectories (ec: system:267 The directory name is invalid.)
5: C:\J\workspace\ign_fuel-tools-gz-8-win\ws\gz-fuel-tools\src\FuelClient_TEST.cc(46): error: Value of: common::createDirectories( common::joinPaths(modelPath, "2", "meshes"))
5:   Actual: false
5: Expected: true
```

</details>

mjcarroll, 2023-10-30:
Closed by #376

azeey, 2023-11-06:
Let's disable the tests on Windows for gz-fuel-tools8.

Crola1702, 2023-11-14:
This problem is still happening in gz-fuel-tools8 windows consistently ([Reference build](https://build.osrfoundation.org/job/ign_fuel-tools-gz-8-win/91/#showFailuresLink))

mjcarroll, 2023-12-11:
Closed by #387

closed 2023-12-11 by mjcarroll