gazebosim/gz-fuel-tools #420: Gz fuel server allows models with duplicate author and name

luca-della-vedova, 2024-04-30:
It seems two models with exactly the same name and author made it into the gazebo app portal:

[image]

Clicking each of them points to the same page https://app.gazebosim.org/anni/models/ur5_rg2.

This is causing a fair bit of issue in the Open-RMF side of things where we use model names to build a cache of fuel models and I believe it is probably not indended behavior?

azeey, 2024-04-30:
@luca-della-vedova can you verify if this is still a problem. There was a caching issue on the Fuel server that was resolved last week which might have also fixed this. The [URL](https://app.gazebosim.org/anni/models/ur5_rg2) for the ur5 model in the description no longer works for example.

azeey, 2024-04-30:
However, I do see that the model count is 2 when [searching](https://app.gazebosim.org/search;q=ur5) for ur5 even though there's only one model shown 
[image]

luca-della-vedova, 2024-05-02:
The way I'm using to verify the issue is to look at the [models page](https://app.gazebosim.org/fuel/models), since the model has been uploaded fairly recently it appears in the page as one of the latest ones:

[image]

azeey, 2024-05-02:
It looks like listing the model of just that owner shows there's one model, but if I list all models, it shows two `ur5_rg2` models owned by `anni`.
```
$ gz fuel list -v4 -t model -o anni
Fetching model list from https://fuel.gazebosim.org...
Received model list (took 5ms).
https://fuel.gazebosim.org
└── anni
    └── ur5_rg2
1 owners, 1 models

```

```
$ gz fuel list -v4 -t model 
Fetching model list from https://fuel.gazebosim.org...
Received model list (took 57653ms).
https://fuel.gazebosim.org
...
├── anni
│   ├── ur5_rg2
│   └── ur5_rg2
```

cc @nkoenig