gazebosim/gz-fuel-tools #455: Spell check the documentation

ahcorde, 2025-03-18:
As part of our spring cleaning, we want to detect spell issues in our documentation and fix it when possible. Here's a potential plan to tackle this issue:

1. We suggest to start checking out the main branch.
1. Identify the files that contain documentation. These are typically `.md` (Markdown) or `.md.in` (a Markdown file to be generated by `CMake`).
1. Install in your system a tool to check for spelling errors (e.g.: `aspell` in Ubuntu).
1. Run the spell checks in all the documents. Most of the catches will be false positives due to words like `Protobuf`, `CMake` and other words that aren't typically in an English dictionary. Ignore all of these until you catch a real spell problem.
1. Fix it. Usually these programs recommend a proper word or you can later manually modify the file with the correct English term.
1. Create a pull request with your fixes. [This guide](https://gazebosim.org/docs/all/contributing/#contributing-code) can be helpful. Don't be shy, we'll be kind during the process and help you if you don't have much experience with Gazebo yet or the instructions are not clear enough.

khaledgabr77, 2025-03-26:
I'll take care of this issue, but I have a question: For technical documentation that includes a lot of code, is `camelCase` the recommended format?

khaledgabr77, 2025-03-29:
Please check out PR #456.

ahcorde, 2025-03-31:
Thank you @khaledgabr77, 

I can see some more issues:

```bash
./include/gz/fuel_tools/FuelClient.hh:303: depedencies ==> dependencies
./include/gz/fuel_tools/FuelClient.hh:463: identifer ==> identifier
./include/gz/fuel_tools/FuelClient.hh:486: inout ==> input, in out
./include/gz/fuel_tools/WorldIter.hh:71: Derefence ==> Dereference
./include/gz/fuel_tools/Interface.hh:55: withing ==> within
./include/gz/fuel_tools/Interface.hh:59: coult ==> could
./test/gtest_vendor/include/gtest/internal/gtest-param-util.h:756: susequently ==> subsequently
./test/gtest_vendor/include/gtest/internal/gtest-param-util.h:767: repored ==> reported, reposed
./test/gtest_vendor/include/gtest/internal/gtest-death-test-internal.h:195: SEH ==> SHE
./test/gtest_vendor/include/gtest/internal/gtest-port.h:711: SEH ==> SHE
./test/gtest_vendor/include/gtest/internal/gtest-port.h:714: SEH ==> SHE
./test/gtest_vendor/src/gtest-internal-inl.h:389: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:841: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:844: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:847: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:850: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2536: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2574: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2575: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2576: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2576: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2578: SEH ==> SHE
./test/gtest_vendor/src/gtest.cc:2605: SEH ==> SHE
./tools/gz-update-assets.py:17: case-insentive ==> case-insensitive
./src/RestClient.cc:79: Niether ==> Neither
./src/Result_TEST.cc:102: successfull ==> successful
./src/ClientConfig.cc:181: delimeters ==> delimiters
./src/gz.cc:586: sepcified ==> specified
./src/FuelClient.cc:786: depdencies ==> dependencies
./src/FuelClient.cc:1730: confg ==> config
./src/LocalCache.cc:87: tha ==> than, that, the
./src/cmd/cmdfuel.rb.in:52: prefered ==> preferred
./src/cmd/cmdfuel.rb.in:171: confg ==> config
```

khaledgabr77, 2025-03-31:
Thank you @ahcorde 
I initially thought you only wanted to check the documentation files. I’ll go ahead and create a new PR and check the code as well.

khaledgabr77, 2025-04-01:
> ./test/gtest_vendor/include/gtest/internal/gtest-death-test-internal.h:195: SEH ==> SHE
> ./test/gtest_vendor/include/gtest/internal/gtest-port.h:711: SEH ==> SHE [...]

Since `SEH` stands for **Structured Exception Handling** and is a valid technical acronym, I believe it should be added to the `.aspell.en.pws`

khaledgabr77, 2025-04-01:
I fixed code documentation issues here #458 @ahcorde

closed 2025-04-08 by kscottz